ketilmo/balena-ads-b #320: Wingbits error

Placemaster, 2025-04-14:
I have created a Wingbits account according to the setup page. 
Registered first station, linked a wallet.
On the balena dashboard I added a variable named WINGBITS_DEVICE_ID. Put my Id in.
Under services I did cycle Wingbits.

After cycling (and a complete reboot) I do get the following log messages:

` wingbits  Verifying settings...
 wingbits   
 wingbits  Wingbits Device ID is set: Nijmegen-Radar
 wingbits  Receiver host is set: dump1090-fa
 wingbits  Receiver port is set: 30005
 wingbits   
 wingbits  Settings verified, proceeding with startup.
 wingbits   
 wingbits  [readsb-wingbits]     invoked by: /usr/bin/feed-wingbits --net --net-only --debug=n --quiet --net-connector localhost,30006,json_out --write-json /run/wingbits-feed --net-beast-reduce-interval 0.5 --net-heartbeat 60 --net-ro-size 1280 --net-ro-interval 0.2 --net-ro-port 0 --net-sbs-port 0 --net-bi-port 30154 --net-bo-port 0 --net-ri-port 0 --net-connector dump1090-fa,30005,beast_in
 wingbits  [readsb-wingbits]     Mon Apr 14 19:29:16 2025 UTC  readsb starting up.
 wingbits  [readsb-wingbits]     readsb version: 3.14.1600 wiedehopf git: f535e51 (committed: Tue Mar 14 13:17:07 2023 0100)
 wingbits  [readsb-wingbits]     30154: Beast TCP input port           
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Beast TCP input: Attempting connection to dump1090-fa port 30005 ...
 wingbits  [readsb-wingbits]     Beast TCP input: Connection established: dump1090-fa (172.17.0.3) port 30005
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (127.0.0.1) port 30006 failed: 111 (Connection refused)
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (::1) port 30006 failed: Transport endpoint is not connected
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (127.0.0.1) port 30006 failed: 111 (Connection refused)
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (::1) port 30006 failed: Transport endpoint is not connected
 wingbits  2025-04-14T19:29:17.664942Z  INFO vector::app: Log level is enabled. level="vector=info,codec=info,vrl=info,file_source=info,tower_limit=info,rdkafka=info,buffers=info,lapin=info,kube=info"
 wingbits  2025-04-14T19:29:17.801179Z  INFO vector::config::watcher: Creating configuration file watcher.
 wingbits  2025-04-14T19:29:17.813408Z  INFO vector::config::watcher: Watching configuration files.
 wingbits  2025-04-14T19:29:17.831954Z  INFO vector::app: Loading configs. paths=["/etc/vector/vector.yaml"]
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (127.0.0.1) port 30006 failed: 111 (Connection refused)
 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...
 wingbits  [readsb-wingbits]     Position json output: Connection to localhost (::1) port 30006 failed: Transport endpoint is not connected

 wingbits  2025-04-14T19:29:19.126048Z  INFO vector::topology::running: Running healthchecks.

 wingbits  2025-04-14T19:29:19.133435Z  INFO vector: Vector has started. debug="false" version="0.34.2" arch="arm" revision="d685a16 2024-01-02 14:59:54.890517436"
 
wingbits  2025-04-14T19:29:19.133607Z  INFO vector::app: API is disabled, enable by setting `api.enabled` to `true` and use commands like `vector top`.
 
wingbits  2025-04-14T19:29:19.133841Z  INFO source{component_kind="source" component_id=readsb component_type=socket}: vector::sources::util::net::tcp: Listening. addr=127.0.0.1:30006

 wingbits  [readsb-wingbits]     Position json output: Attempting connection to localhost port 30006 ...

 wingbits  [readsb-wingbits]     Position json output: Connection established: localhost (127.0.0.1) port 30006

 wingbits  2025-04-14T19:29:29.127377Z ERROR vector::topology::builder: msg="Healthcheck timed out." component_kind="sink" component_type="vector" component_id=vector

 wingbits  2025-04-14T19:29:29.142970Z  WARN sink{component_kind="sink" component_id=vector component_type=vector}:request{request_id=1}: vector::sinks::util::retries: Retrying after error. error=Request failed: status: Unknown, message: "operation was canceled: request has been canceled", details: [], metadata: MetadataMap { headers: {} } internal_log_rate_limit=true

 wingbits  2025-04-14T19:30:30.146595Z  WARN sink{component_kind="sink" component_id=vector component_type=vector}:request{request_id=1}: vector::sinks::util::retries: Request timed out. If this happens often while the events are actually reaching their destination, try decreasing `batch.max_bytes` and/or using `compression` if applicable. Alternatively `request.timeout_secs` can be increased. internal_log_rate_limit=true
 `

What can I do to resolve the problem?
All other services are running without an issue (already for years).

shawaj, 2025-04-18:
@Placemaster you need to update your balena-ads-b setup using the instructions here - https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-16--updating-to-the-latest-version

You are running a version which is very old and still uses vector (this was changed around 4 months ago)

Once you update it should work

shawaj, 2025-04-19:
@Placemaster I also just realised, I don't think you followed the setup instructions correctly here https://github.com/ketilmo/balena-ads-b?tab=readme-ov-file#part-9--configure-wingbits

The `WINGBITS_DEVICE_ID` is not something you can just create yourself. It comes from the Wingbits dashboard in the URL of the station.

You also need a GeoSigner. Do you have one?

Placemaster, 2025-04-19:
I have updated the system to the latest version and now it has an heartbeat. I do not have the Geosigner yet. I will kale sure I will get one.

While creating an account on Wingbits you are able to choose your own name (it will be stores as an alias). Underneath you will still get the 3 word name.

closed 2025-04-19 by Placemaster

shawaj, 2025-04-19:
Awesome... Glad it's working now for you 🥳